T-vK/ESP32-BLE-Mouse #1: Compiles, but is not working as expected.

duke2421, 2019-08-06:
Hi,

I followed your work already at th ESP-BLE-Snippets. 
The basic Arduino code from chegewara is sending the text. 
Now I tried your library but can't get it to work.
Here is my test code:

```
#include <BleMouse.h>

BleMouse bleMouse;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleMouse.begin();
  pinMode(12, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(12), clickNumLock, FALLING);   // Num Lock
}

void loop() {
}

IRAM_ATTR void clickNumLock(){
  if(bleMouse.isConnected()) {
    bleMouse.click(MOUSE_BACK);
    Serial.println("Buttonpressed");
  }
} 
```

It compiles with no errors, Win10 and Android are able to connect, but under Win10 nothing happens, with Android the mouse pointer appears and moves a little to the right after each keystroke, no matter which command is sent (MOUSE_LEFT, RIGHT or BACK). Am I doing it all wrong?

btw. do you think, you can include 3 more functions in your library. My goal is to make a simple media controll button with a rotary encoder and with following functions: "play/pause", "Volume up", "Volume down" (eventually "next song"). But I am not able to send these keystrokes over BLE with my ESP32, my coding skills are by far not enough ;).

best regards
Daniel

T-vK, 2019-08-06:
Damn, I can confirm it doesn't work properly in Android. (I can't test it on Windows right now.)
I tested every feature on a Linux computer before releasing v0.1 and everything worked fine there.
This is a pretty weird bug, I'm not sure why this is happening.

Maybe @chegewara has an idea.

About your request to include keyboard features in this library: This is a mouse library so it should only have mouse related functionality. Maybe I'll make an attempt to create a keyboard library which then obviously would be able to send media keys like "play/pause" "volume up/down" etc. 
But at the time being I'm apparently not even able to write a ble mouse library that works properly. 
I wouldn't be surprised if writing a ble keyboard library would be orders of magnitudes more difficult.

T-vK, 2019-08-07:
@duke2421 Can you test this release please: https://github.com/T-vK/ESP32-BLE-Mouse/releases/tag/tmp

It might fix the issues, but I'm not sure.

duke2421, 2019-08-07:
Hi @T-vK, I tested the tmp release. No change in behavior, sorry. 
With your new HID device descriptor the compiling gives an error message, INPUT was not declared, so I changed INPUT(1) to HIDINPUT(1) and then it is working on Android. I will test it on Win10 later today and then come back.

About my project to create a media controller, I agree, it should not be in the mouse library. But now I have a working sample and can test some more things.
Could you give me a few hints, where you found more explanation about the HID device descriptor? I think, if I understand that piece of code a bit better, I should be able to create a working code for my project.

duke2421, 2019-08-07:
@T-vK I tested it on Win10 and also there it is working with your new HID device descriptor.

T-vK, 2019-08-07:
@duke2421 That's great to hear! I'll push the fix later today and create another release.

This helped me a lot understanding HID device descriptors:
https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/
https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/#comment-2229
https://www.freebsddiary.org/APC/usb_hid_usages.php
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HIDTypes.h

If I can find the time later today I'll try to make a ble keyboard library or at least check how difficult it would be.

duke2421, 2019-08-07:
@T-vK Don't worry, there's no hurry.

Thanks for the links.

T-vK, 2019-08-07:
@duke2421 I have started making a keyboard library: https://github.com/T-vK/ESP32-BLE-Keyboard

I haven't tested it all all though, so it might not work.

chegewara, 2019-08-07:
> @duke2421 That's great to hear! I'll push the fix later today and create another release.
> 
> You are absolutely right about using `HIDINPUT` instead of `INPUT`, looking at HIDTypes.h:
> 
> ```
> #ifdef ARDUINO_ARCH_ESP32
> #define HIDINPUT(size)             (0x80 | size)
> #define HIDOUTPUT(size)            (0x90 | size)
> #else
> #define INPUT(size)             (0x80 | size)
> #define OUTPUT(size)            (0x90 | size)
> ```
> 
> This helped me a lot understanding HID device descriptors:
> https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/
> https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/#comment-2229
> https://www.freebsddiary.org/APC/usb_hid_usages.php
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HIDTypes.h
> 
> If I can find the time later today I'll try to make a ble keyboard library or at least check how difficult it would be.

Fell free to create PR and change INPUT to HIDINPUT and OUTPUT to HIDOUTPUT.

T-vK, 2019-08-08:
This should be fixed now in [v0.2.0](https://github.com/T-vK/ESP32-BLE-Mouse/releases/tag/v0.2.0).

closed 2019-08-08 by T-vK

filippoitaliano, 2022-06-19:
I have the same problem on Win11. It compiles but it doesn't actuate mouse commands. BLE Keyboard works indeed. I've also tested the BLE Combo lib and it works entirely but the mouse commands. I'm working with an AZDelivery esp32 dev board. This is my testing code:

```
#include <BleMouse.h>

BleMouse bleMouse;

int KEY_PIN_5 = 27;
int KEY_STATE_5 = HIGH;
int KEY_PREV_STATE_5 = HIGH;

int KEY_PIN_6 = 12;
int KEY_STATE_6 = HIGH;
int KEY_PREV_STATE_6 = HIGH;

void setup() {
  pinMode(KEY_PIN_5, INPUT_PULLUP);
  pinMode(KEY_PIN_6, INPUT_PULLUP);
  Serial.begin(115200);
  bleMouse.begin();
}

void loop() {
  if (bleMouse.isConnected()) {

    KEY_STATE_5 = digitalRead(KEY_PIN_5);
    if (KEY_STATE_5 == LOW && KEY_PREV_STATE_5 == HIGH) {
        Serial.println("key 5 pressed");  
        bleMouse.click(MOUSE_LEFT);
        delay(100);
    }
    KEY_PREV_STATE_5 = KEY_STATE_5;

    KEY_STATE_6 = digitalRead(KEY_PIN_6);
    if (KEY_STATE_6 == LOW && KEY_PREV_STATE_6 == HIGH) {
        Serial.println("key 6 pressed");  
        bleMouse.click(MOUSE_RIGHT);
        delay(100);
    }
    KEY_PREV_STATE_6 = KEY_STATE_6;
    
  } else {
    Serial.println("Not connected");
    delay(2000);
  }
}
```