T-vK/ESP32-BLE-Mouse #2: ESP32-BLE-Gamepad

lemmingDev, 2019-08-09:
Thanks heaps for making ESP32-BLE-Mouse and ESP32-BLE-Keyboard
It will make it much easier to use that functionality now

Am hoping you may consider creating ESP32-BLE-Gamepad

A ton of possibilities....

T-vK, 2019-08-09:
First I'll have to finish the ESP32-BLE-Keyboard library. I'll see if I feel like making another one then.

lemmingDev, 2019-08-10:
Thanks for considering it. Would be fantastic

T-vK, 2019-08-13:
This could be pretty helpful: https://xboxdevwiki.net/Xbox_Input_Devices
Seems relatively straight forward, if I wanted to create an Xbox controller library. 
But it would be much nicer to have a general library though that supports all kinds of gamepads.
That would require a lot of research though. Basically I'd need to know how the HID device descriptors of the most common HID gamepads differ.

lemmingDev, 2019-08-13:
In my humble opinion, an Xbox 360 / One controller layout would be a fantastic start.

It is widely supported and is what 99% of users of such a library would want.

lemmingDev, 2019-09-12:
Not sure ho much work it would take to port this over, but it's really flexible and seems to do what you commented on earlier; "would be much nicer to have a general library though that supports all kinds of gamepads"

lemmingDev, 2019-09-14:
After one night doing research and another (late one) implementing and troubleshooting, I was able to modify your ESP32-BLE-Mouse library into ESP32-BLE-Gamepad

[ESP32-BLE-Gamepad.zip](https://github.com/T-vK/ESP32-BLE-Mouse/files/3612195/ESP32-BLE-Gamepad.zip)

Learnt heaps!

Has 14 buttons (1~14)

Has 6 Axis (x, y, z, rZ, rX, rY) --> (Left Thumb X, Left Thumb Y, Right Thumb X, Right Thumb Y, Left Trigger, Right Trigger)

The L/R triggers are separate, as in the PS controllers, vs the combined ones in the XBOX controllers.
Might change it if I can ever work out how to make the computer think its an XBOX controller by default. Can the hid->pnp and   hid->hidInfo hex values be used to achieve this?

Still needs some polish such as adding the point of view hat (d-pad), and using a struct? for the report so I can use values other than 8 bit signed chars in the array currently used.

I put the left/right triggers separate to the other axes in the report as I want to make them 0~256 instead of -127~127, but I need to sort out that last point first.

Thanks heaps for the mouse library.

BTW - If you'd like to host the library on your Github, feel free - I don't need any attribution

lemmingDev, 2019-09-14:
DPAD was easier than I thought - here is the new version of the library that supports it.

[ESP32-BLE-Gamepad.zip](https://github.com/T-vK/ESP32-BLE-Mouse/files/3612284/ESP32-BLE-Gamepad.zip)

So - its pretty functional now

T-vK, 2019-09-14:
That's awesome!
Unless you are planning to abandon your Github account, I think you should make your own repository for the library. This way it would be much easier for you make changes to it. 

To make the computer think it's an Xbox controller, it might be enough to use the same device name and manufacturer as the original one. The HID device descriptor would ideally be equal to the original one as well. If you have such a controller lying around, you might be able to extract it using the appropriate software. Otherwise it can probably be found somewhere on the Internet.

lemmingDev, 2019-09-14:
OK

My first ever Github repository

[https://github.com/lemmingDev/ESP32-BLE-Gamepad](https://github.com/lemmingDev/ESP32-BLE-Gamepad)

T-vK, 2020-10-21:
I will close this issue as I don't think it's needed anymore.

closed 2020-10-21 by T-vK