T-vK/ESP32-BLE-Mouse #3: IoS 13 compatibility?

00ghost, 2019-10-07:
Hi,

I would like to use the new feature in IoS 13 to use an external mouse and tried your lib. However, it seems that ESP32-BLE-Mouse does not work for me to an iPhone 7 with IoS 13.1.2.  I e, the ESP connects to the phone but then nothing. Same library work fine to a MacBookPro with MacOS 10.14.6. The ESP32-BLE-Keyboard library works fine (but only a few keys tested) with both these devices. Accessibility settings on the phone seem to be correct. ESP used is a Sparkfun ESP32 Thing.
Anything I can test or try?

Göran

T-vK, 2019-10-07:
Unfortunately there is probably nothing you can do. Your phone most likely only supports normal Bluetooth mice, but not Bluetooth LE ones. 
The same issue has already been reported for the ble keyboard library btw: https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2

lawleo, 2019-11-30:
https://github.com/asterics/esp32_mouse_keyboard
I can successfully connect my iPad 6th gen (iOS 13) with about program.  It requires esp-idf v3.3 (not 4.x) to compile.  Can anyone find out the reason why that works but not this project?

T-vK, 2019-11-30:
I'd be very surprised if the 6th gen iPad really doesn't work with this library. At least people have reported that my keyboard library (which uses almost the exact same code as this mouse library for the Bluetooth communication), works fine on the following iPads: iPad Air2, iPad Pro 2, the lastest iPad Air

Can you confirm that you can connect your ESP32 with other devices (e.g. Android, Windows, OS X ) successfully?

lawleo, 2019-12-01:
Yes, my code work in Android & MacOS 10.14 too (sorry, no windows), but it does not work in my iPad 6th gen.  my ipad can discover the mouse and connect to it, but can't move on any direction or click the left button.

shrimpchip28, 2020-01-30:
I just tested this works great and detected on my samsung galaxy note 9. Tested on an iphone 6s with iOS 13.3. As mentioned, phone detects my device but after going through the required setting changes on the phone, nothing happens.  To confirm and  test general mouse support, i paired the same iPhone with my logitech MX Master 2 mouse and that worked.

T-vK, 2020-01-30:
It's really hard to say what the problem is here. An incomplete or buggy BLE implementation on Apple's side seems most likely to me (be it a hardware-, firmware- or driver issue). Your mouse is probably not a BLE mouse or it has a fallback to support the older non-BLE protocol. At least that's my guess...

shrimpchip28, 2020-01-31:
> It's really hard to say what the problem is here. An incomplete or buggy BLE implementation on Apple's side seems most likely to me (be it a hardware-, firmware- or driver issue). I mean it seems to work on Android, Windows, MacOS, some iPhones and some iPads... Your mouse is probably not a BLE mouse or it has a fallback to support the older non-BLE protocol. At least that's my guess...

Thanks.  Hmmm so here's the requirements for the Logitech MX Master 2s that I have. Indicates BLE is required (unless of course you use the Logitech Unifier dongle).  https://www.logitech.com/en-roeu/product/mx-master-2s-flow

Will try and check out another person's iPad.

nori-dev-akg, 2020-05-02:
delete line 28 in BleMouse.cpp.
`//  REPORT_ID(1),        0x01, //     REPORT_ID (1)
`

seigel, 2020-06-01:
@nori-dev-akg Does that help?

milkai, 2020-06-05:
> delete line 28 in BleMouse.cpp.
> `// REPORT_ID(1), 0x01, // REPORT_ID (1) `

This worked

seigel, 2020-06-05:
Alright...I'll have to give it a go!  Thx

AiueoABC, 2020-06-10:
This worked for me too! Thank you so much.

lawleo, 2020-06-15:
I can confirm this work for me too! Thanks a lot.

consciouspnm, 2020-06-15:
That works for me as well, iPad Air 2, IOS 13.5.1

xcarcelle, 2020-09-20:
Hi there,
I have been testing lately with iOS13.7 (iPAD Pro 10.5) and if turning the esp32 module then the reconnexion is not automatic and we need to "forget the device" in the BT settings on the ipad and reconnect the BLE device in order to properly work again.
Does anyone have experienced this issue ?
Much regards, Xavier.

consciouspnm, 2020-09-25:
IOS 14.0 seems to reconnect ok. If I turn the esp32 off it says not connected, when I turn it back on it connects and the pointer re-appears

T-vK, 2020-09-25:
Has anyone of you removed line 28 as mentioned [here](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issuecomment-622987303) by nori-dev-akg and then tested if it then still works on Windows, Linux or Android?

consciouspnm, 2020-09-25:
Just tested on my Surface Pro and it doesn't work.  So, you need to delete the line for IOS and keep it for Windows.  Windows will see it and pair but the pointer won't move now.

consciouspnm, 2020-09-27:
@T-vK Your descriptor (uint8_t _hidReportDescriptor) uses constants i.e. 
USAGE_PAGE(1),       0x01, // USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x02, // USAGE (Mouse)
  COLLECTION(1),       0x01, // COLLECTION (Application)
  USAGE(1),            0x01, //   USAGE (Pointer)
The ones on the web just use hex i.e.
0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
0x09, 0x02,                    // USAGE (Mouse)
0xa1, 0x01,                    // COLLECTION (Application)
0x09, 0x01,                    //   USAGE (Pointer)
If I replace the constants with a hex version from the web, it compiles but doesn't work.  How do I get blemouse.cpp to accept hex descriptors e.g. the Standard Mouse one from here: https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

T-vK, 2020-09-27:
Those macros are defined here: 

https://github.com/nkolban/esp32-snippets/blob/fe3d318acddf87c6918944f24e8b899d63c816dd/cpp_utils/HIDTypes.h#L60

Since `0x04 | 1` == `0x05`, `USAGE_PAGE(1)` should be equal to `0x05`.  
But it really doesn't make a difference if you use those macros or hex values directly.  
From what I can tell the two snippets you provided are equal. Maybe you made a mistake somewhere else in your code.

consciouspnm, 2020-09-27:
@T-vK The difference are further down the code, so I copied and pasted the full set from the above link into the descriptor as this should be the minimum needed for a standard mouse.  Nothing else was changed but it doesn't work on Windows or IOS:

0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
0x09, 0x02,                    // USAGE (Mouse)
0xa1, 0x01,                    // COLLECTION (Application)
0x09, 0x01,                    //   USAGE (Pointer)
0xa1, 0x00,                    //   COLLECTION (Physical)
0x05, 0x09,                    //     USAGE_PAGE (Button)
0x19, 0x01,                    //     USAGE_MINIMUM (Button 1)
0x29, 0x03,                    //     USAGE_MAXIMUM (Button 3)
0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
0x25, 0x01,                    //     LOGICAL_MAXIMUM (1)
0x95, 0x03,                    //     REPORT_COUNT (3)
0x75, 0x01,                    //     REPORT_SIZE (1)
0x81, 0x02,                    //     INPUT (Data,Var,Abs)
0x95, 0x01,                    //     REPORT_COUNT (1)
0x75, 0x05,                    //     REPORT_SIZE (5)
0x81, 0x03,                    //     INPUT (Cnst,Var,Abs)
0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
0x09, 0x30,                    //     USAGE (X)
0x09, 0x31,                    //     USAGE (Y)
0x15, 0x81,                    //     LOGICAL_MINIMUM (-127)
0x25, 0x7f,                    //     LOGICAL_MAXIMUM (127)
0x75, 0x08,                    //     REPORT_SIZE (8)
0x95, 0x02,                    //     REPORT_COUNT (2)
0x81, 0x06,                    //     INPUT (Data,Var,Rel)
0xc0,                          //   END_COLLECTION
0xc0                           // END_COLLECTION

T-vK, 2020-09-27:
It's been too long for me, so I can't quite remember what exactly a valid device descriptor would look like. But the most obvious differences that I see are:
- no REPORT_ID (1)
- first and second occurrence of REPORT_COUNT and  REPORT_SIZE are swapped 
- no USAGE (Wheel)

But I'm not sure if any of these are the actual problem. From my experience I can tell you though, that most of the descriptors that you'll find online are not valid and even if you have a perfectly valid one, that is by no means a guarantee that it'll work on all operating systems.

consciouspnm, 2020-09-28:
@T-vK I've now got it working with IOS and Windows using your descriptor. The REPORT_ID is invalid as it is only used for multiple devices like a mouse and keyboard so that line needs deleting.  Doing that means this line of code is no longer valid:

  `bleMouseInstance->inputMouse = bleMouseInstance->hid->inputReport(1); // <-- input REPORTID from report map`

so it needs changing to:

  `bleMouseInstance->inputMouse = bleMouseInstance->hid->inputReport(0); // <-- input REPORTID from report map`

I've tested it on IOS 13.7 and 14 using iPhones and iPads and a Surface Pro and a ThinkPad with a BT 5.0 dongle.
I still don't know why the standard mouse descriptor doesn't work, even the one from usb.org generated using their tool, unless there is something in your code which is looking for something in your descriptor.

T-vK, 2020-09-28:
If you'd like to make a Pull Request I'd be down to accept it.

T-vK, 2020-10-20:
This has been fixed in the new release (0.3.1). I hope I did everything as described. 
I'll close this issue for now.
If anyone with an iOS 13 device still encounters issues, tell me and I will reopen this issue.

closed 2020-10-20 by T-vK

T-vK, 2020-10-21:
Hey everyone, I am trying to make a list of apple devices that work or don't work with this library.
If you read this and you have an Apple device that you have tested, I would very much appreciate it if you could add a comment in #20. (Unless of course it is already listed in there.)

ShiverZm, 2020-11-08:
mark this issure ,my iphone did not  work