T-vK/ESP32-BLE-Mouse #6: Doesn't appear to work on windows 7

AlessandroAU, 2020-04-07:
I've uploaded an example, I can pair using two different windows 10 laptops and my android phone. However, I tested on two other windows 7 machines and they cannot see the Bluetooth device at all.

T-vK, 2020-04-07:
Maybe your Windows 7 machines don't support Bluetooth LE?

AlessandroAU, 2020-04-07:
Hmm interesting, I am using a CSR 4.0 dongle (CSR8510) which should support LE

T-vK, 2020-04-07:
Maybe your driver doesn't (fully) support BLE. If you're using a driver provided automatically by Windows, you might wanna install the driver the manufacturer provides if they do.

T-vK, 2020-10-21:
Can you try again with the latest release? I don't have high hopes, but maybe it resolves your issue.