T-vK/ESP32-BLE-Mouse #7: Windows 10 connection problems (solved)

joseluu, 2020-04-15:
To make it connect on Windows 10, I had to do the following:

replace the first line below by the next 3
  //pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);
  pSecurity->setCapability(ESP_IO_CAP_OUT);
  pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

I am not knowledgeable on BLE, I cannot make a properly documented pull request.

zack-cpp, 2020-12-09:
in which library?

T-vK, 2020-12-10:
Thank you for the information. 
Can you tell me which Bluetooth adapter you've used? I'm asking because other people have reported that it works fine on Windows 10 the way it was, but maybe this is different depending on which Bluetooth card/dongle you're using.

joseluu, 2020-12-10:
The adapter is the one integrated in a HP elitebook X3U21AV fairly recent,
the official spec does not say much about it unfortunately
https://support.hp.com/ie-en/document/c05355646
In the device manager it looks like to be part of the Intel chipset
[image: image.png]