T-vK/ESP32-BLE-Mouse #16: Does it control iphone?

ShiverZm, 2020-08-31:
as title is , it means use arduino control iphone?

T-vK, 2020-08-31:
I don't have any Apple devices, so I can't really support them.
It seems to work for some iPhones if you read through the issues.

consciouspnm, 2020-09-25:
I have an iPhone XS and an iPad Air both with IOS 14 and it does work.

ShiverZm, 2020-09-25:
cooool,does it  need jailbroken?

DareDareHub, 2020-09-25:
@consciouspnm did you do anything special to make it work ? 
It doesn't work on my iPAD 10.5" iOS 14 (doesn't even appear in the the bluetooth devices list) 
nor on my iPhone SE iOS13.4 (appears and connects in Bluetooth and AssistiveTouch devices BUT the pointer doesn't move)
I'm using an ESP-32 DevkitD v4...

consciouspnm, 2020-09-25:
Did you remove the REPORT_ID line mentioned in the other thread?

DareDareHub, 2020-09-28:
@consciouspnm 
Thanks, yes I've removed the REPORT_ID...
and thanks, Indeed Assistive touch doesn't need to be turned on ! (I'm connecting a BLE mouth successfully)
but still no luck with the ESP-32 and iOS 14, would you please  share your descriptor code so I can test and feedback on the results. maybe create a table of compatible hardware.

consciouspnm, 2020-09-28:
@DareDareHub Yes, if you change anything it's best to forget the device and re-add it.

DareDareHub, 2020-09-29:
All that's left to find now is absolute mouse ! :)

T-vK, 2020-10-21:
With the latest release there is no need to remove the REPORT_ID anymore.
Also @DareDareHub Take a look at #15 and #5. There are some examples. They aren't perfect, but maybe they do the job for you.

T-vK, 2020-10-21:
I am trying to make a list of apple devices that work or don't work with this library. 
If you read this and you have an Apple device that you have tested, I would very much appreciate it if you could add a comment in #20. (Unless of course it is already listed in there.)

ShiverZm, 2020-11-04:
maybe iphone 8 plus did not work ,i use your report descriptor.

consciouspnm, 2020-11-06:
@ShiverZm What version of IOS is it?

ShiverZm, 2020-11-06:
IOS　１４．１

ShiverZm, 2020-11-06:
but i  used relative mouse report descriptor ,that worked, but it seems not convenient,some method can help?

ShiverZm, 2020-11-10:
somebody knew, how move mouse without press left button?

ShiverZm, 2020-11-12:
@consciouspnm what do you think?bro

consciouspnm, 2020-11-12:
@ShiverZm You've got something wrong in the code if you have to press the button.  Also, the iPhone 8 should work as it supports IOS14 and Bluetooth LE.  Have you tried your code and hardware on anything else?
The move code is:
```
BLEMouse.move(2, 2, 0, 0);
delay(10);
```
a click is:
`BLEMouse.click(1);`

ShiverZm, 2020-11-20:
i do so,but when i move  for a constant time , it  clicked  not move ,it just came up for my iphone 8 plus ,my air2 is ok

ShiverZm, 2020-11-20:
(2,2,0,0) it means reportId= 2, ButtonType = 2 (right Button), moveX= 0,moveY =0 ?right?

ShiverZm, 2020-11-20:
@consciouspnm

ShiverZm, 2020-11-20:
And did you know the Iphone Menu button how to send? i use IPAD send ButtonType = 5 is Menu Function,but it did not work for iphone

ShiverZm, 2020-11-20:
@consciouspnm  my fault, ButtonType = 5 is Home Button

ShiverZm, 2020-11-23:
@consciouspnm would you mind giving me your email?maybe it will be convinent to talk