T-vK/ESP32-BLE-Mouse #18: Landscape orientation issue with Android absolute mouse mode

xcarcelle, 2020-10-17:
Hi there,
after running this code [esp32-ble-mouse-abs-android.ino.txt](https://github.com/T-vK/ESP32-BLE-Mouse/files/5395456/esp32-ble-mouse-abs-android.ino.txt) with Android 8/9, it enables correctly the absolute mode when moveTo(x, y) goes directly to the target position on Android devices. However this works in portrait mode but in landscape mode, we observed that the virtual enabled zone is still in portrait mode and the mouse cannot go beyond the width of the portrait mode.
It seems difficult to determine if it is due to Android in rotation mode to landscape or some other issue with the HID descriptor and send report ? Have you guys experienced such an issue with Android device in absolute mode ? 
Devices tested : tablets (Lenovo and Huawey models along with some android phones) in Android 8.0/9.0.
Cheers.

T-vK, 2020-10-17:
Your code uses an entirely different report descriptor than this library. I have no experience with  digitizer input. Seems like other people have similar problems with digitizer input: https://stackoverflow.com/questions/28536602/hid-digitizer-descriptor-doesnt-perform-well-with-landscape-orientation
As long as your code is aware of the screen resolution, orientation, cursor speed and acceleration, you could make it work with a mouse report descriptor.

xcarcelle, 2020-10-17:
@T-vK : have you observed orientation issue with Android devices and absolute mouse mode ?
How would tell the code about the orientation ? Do you think that could be sufficient for the Android kernel to be aware of the landscape mode ?

T-vK, 2020-10-18:
I have not tried it. 
The only way might be to have a dedicated piece of software running on the phone that communicates with the ESP.

francoisnicolas, 2023-03-18:
Look at 👍
 0x16, 0x00, 0x00,  //     Logical Minimum (0)
0x36, 0x00, 0x00,  //     Physical Minimum (0)
0x26, 0x80, 0x07,  //     Logical Maximum (1920)     <------
0x46, 0x80, 0x07,  //     Physical Maximum (1920)
0x09, 0x31,        //     Usage (Y)
0x16, 0x00, 0x00,  //     Logical Minimum (0)
0x36, 0x00, 0x00,  //     Physical Minimum (0)
0x26, 0x38, 0x04,  //     Logical Maximum (1080)     <----
0x46, 0x38, 0x04,  //     Physical Maximum (1080)

Perhaps, it's the coordiante system

commit 82f254af (2022-01-17): Add Abs Mouse project

Closes #19 
Closes #18 
Closes #5