T-vK/ESP32-BLE-Mouse #20: List of supported / unsupported Apple devices (iPhones, iPads, Macs etc.)

T-vK, 2020-10-21:
I would like to maintain a list of Apple devices here that work or don't work with this library.
If you have tested a device that is not listed with the same OS version yet, please add a comment and tell me which device and operating system (i.e. iOS version or Mac OS version).
If you tested a device that is already listed, but you had different results (e.g. list says it doesn't work, but for you it does work), please add a comment as well. (And don't forget to mention your device name and OS version.)

**Devices that work**

- [x] iPhone XS | iOS 14 | Reported by [consciouspnm](https://github.com/T-vK/ESP32-BLE-Mouse/issues/16#issuecomment-698808805)
- [x] iPad Air | iOS 14 | Reported by [consciouspnm](https://github.com/T-vK/ESP32-BLE-Mouse/issues/16#issuecomment-698808805)
- [x] iPad Air 2 | iOS 13.5.1 | Reported by [consciouspnm](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issuecomment-644116290)
- [x] iPad Pro 10.5 | iOS13.7 | Reported by [xcarcelle](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issuecomment-695811410) | (Some reconnect issues)
- [x] MacBook Pro (Unknown year) | MacOS 10.14.6 | Reported by [00ghost](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issue-503602025)

**Devices that don't work**

- [ ] iPhone 7 | iOS 13.1.2 | Reported by [00ghost](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issue-503602025)
- [ ] iPad 6th Gen | iOS 13 | Reported by [lawleo](https://github.com/T-vK/ESP32-BLE-Mouse/issues/3#issuecomment-559965477)

xcarcelle, 2020-10-22:
@T-vK : thanks a lot for bootstrapping this table of hardware that will be veru useful. Maybe we can add "relative" and "absolute" mode as we would find eventually a way for absolute mouse HID to work with this library at some point. For testers, I recommand to be careful on : 
- if changing the HID Descriptor library code between two tests, forget the device, turn-off-on the BT on host and reconnect with the peripheral
- generic ble peripheral name on multiple devices advertising at the same time in same location. I use the esp32 wifi mac address to generate the ble device name so there is no confusion on advertising. For instance :
```
uint32_t chipId = 0;
String bleDeviceName;
  for(int i=0; i<17; i=i+8) {
    chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
  }
bleDeviceName = "my-device-" + String(chipId);
bleMouse.deviceName = bleDeviceName.c_str();
```
- I will add more devices and possibly we could do the same with an Android devices table of hardware

Cheers

DareDareHub, 2020-10-22:
If I use @xcarcelle 2nd tip above, the library (BLE connection to a tablet) works well with :

[x] iPad 7th Generation | iOS 14.0.1
[x] iPhone SE 2nd Generation | iOS 13.4
[x] iPad Pro 10.5 | iOS 13.7
[x] iPad Pro 10.5 | iOS 10.3.2 (the 127, -127 moves appeared to be  at least 3 times smaller than on the same tab with iOS 13.7)

[x] Huawei MediaPad T5 | Android 8
[x] Lenovo Tab M8 | Android 9

T-vK, 2020-10-22:
@xcarcelle Thanks for all that information. 
Ideally you would be setting the device name during instantiation:
`BleMouse bleMouse("Bluetooth Device Name", "Bluetooth Device Manufacturer");`
instead of 
`BleMouse bleMouse;`

@DareDareHub Thanks for that extensive list. 

Are you saying that it generally doesn't work for you using the default name "ESP32 Bluetooth Mouse"? Or does it only cause problems when you have multiple ESP32 devices running with that same name?

DareDareHub, 2020-10-26:
@T-vK 
Sorry, yes, it only cause problems when I have multiple ESP32 devices running with that same name. (Edited my post above)

T-vK, 2020-10-26:
But that would be a general issue with all Bluetooth devices, right?

DareDareHub, 2020-10-26:
@T-vK 
Good question, I would be inclined to say so too myself, I do not own 2  same devices that have the same name to try though.

But I think it's safe to say that it's not a library problem but more a problem of how iOS stores and refreshes the information of BLE devices that connect to the tablet.

hkg4788, 2020-11-19:
Tested on iphone6s ,8plus,XR,12,12pro, ipad mini5, ipad pro 1st gen. All work.

T-vK, 2020-11-19:
@hkg4788 Thank you very much, is there any chance you can also tell me the iOS versions of your devices?

Vo-Stru, 2021-03-07:
it runs on iPhone 11 Pro and iPhone SE 1st Gen. Both are on iOS 14.4

londonlove, 2021-10-30:
It‘s don't work well in my iphone7plus 、iphone11、and ipad air3. I think something is wrong with my code

p2c2e, 2022-01-15:
Does not work on my Mac.
OS: macOS Monterey 
Hardware: Macbook Pro 2019 - 16" Intel stuff..

Able to scan and connect to the device, but the demos (for e..g the scroll down) does not work. Serial monitor shows the messages printed...
FWIW : I have tried micropython / raw HID device code and having the same issues - leading me to believe there is something special Mac is expecting. Unable to debug using tools like LightBlue since Mac prevents other regular apps for seeing the "Generic Access" service characteristics...