T-vK/ESP32-BLE-Mouse #23: conflict with wifi

leoncoolmoon, 2021-02-03:
I think this library might conflict with wifi.
I put the wifi stuff and BLE stuff in two separate functions. When I comment out any one of them the program running ok. But if not, the program will upload, but the esp32 will reset halfway through starting. In my program wifi starts before BLE. The wifi starts ok, then before any of the BLE mouse libraries been used the chip is crashed. 
I think it is because BLE and wifi share the same antenna.
any chance you can modify the library, so before "begin()" were called, the library will not use any BLE hardware, please?

leoncoolmoon, 2021-02-03:
just put your sample code with the esp32-cam  CameraWebServer code can recreate the error.
code for CameraWebServer.ino: no change to any other code.
`#include "esp_camera.h"
#include <WiFi.h>
#include <BleMouse.h>
//
// WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
//            or another board which has PSRAM enabled
//

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"

const char* ssid = "ddkk";
const char* password = "0425271998";
BleMouse bleMouse;
void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("Starting BLE work!");
  bleMouse.begin();
  Serial.println(" BLE started!");
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  
  if(bleMouse.isConnected()) {

    unsigned long startTime;

   

    Serial.println("Move mouse pointer up");
    startTime = millis();
    while(millis()<startTime+2000) {
      bleMouse.move(0,-1);
      delay(100);
    }
    delay(500);

    Serial.println("Move mouse pointer down");
    startTime = millis();
    while(millis()<startTime+2000) {
      bleMouse.move(0,1);
      delay(100);
    }
    delay(500);

    Serial.println("Move mouse pointer left");
    startTime = millis();
    while(millis()<startTime+2000) {
      bleMouse.move(-1,0);
      delay(100);
    }
    delay(500);

    Serial.println("Move mouse pointer right");
    startTime = millis();
    while(millis()<startTime+2000) {
      bleMouse.move(1,0);
      delay(100);
    }
    delay(500);

  }

  // put your main code here, to run repeatedly:
 // delay(10000);
}`

Serial output

`Starting BLE work!
 BLE started!
.....
WiFi connected
Starting web server on port: '80'
Starting stream server on port: '81'
abort() was called at PC 0x40087b35 on core 1

Backtrace: 0x40096424:0x3ffd0f70 0x40096655:0x3ffd0f90 0x40087b35:0x3ffd0fb0 0x40087c4b:0x3ffd0fd0 0x4011f656:0x3ffd0ff0 0x4011c509:0x3ffd1110 0x4011f6d5:0x3ffd1420 0x40086c52:0x3ffd1450 0x400e8e96:0x3ffd14a0 0x400e90f1:0x3ffd1500 0x400d2ccc:0x3ffd1530 0x400d193d:0x3ffd15e0 0x400da8ab:0x3ffd1680 0x40092b6d:0x3ffd16a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Starting BLE work!
 BLE started!
....
WiFi connected
Starting web server on port: '80'
E (6595) httpd: httpd_server_init: error in creating ctrl socket (23)
Starting stream server on port: '81'
E (6596) httpd: httpd_server_init: error in socket (105)
Camera Ready! Use 'http://192.168.2.124' to connect
`

jellybins, 2022-03-08:
I encountered the same problem ，when i  used the wifi and ble at  same time。

T-vK, 2022-03-08:
The mouse library might just use too much ram for it to work properly together with the Wifi libraries. Reducing the memory footprint would require a refactoring of this library just like to how it has been done with the keyboard library. Maybe someone will do it some day.

Heisenberg2980, 2022-07-13:
> Have you tried the following?
> 
> https://github.com/wakwak-koba/ESP32-NimBLE-Mouse

I am trying to add that library to VS Code but it is not finding it (I think the reason might be that this fork doesn´t have any release). I will try to download it manually and add it to my project in VS Code, but do you think the memory issue is solved in this fork?

Heisenberg2980, 2022-07-13:
@touchgadget IT WORKS!! Using the new repo was a bit of a challenge as it has dependencies on another repo (NimBLE-Arduino), but after adding both repos it works, and the memory error when trying to use wifi and BLE at the same time doesn´t happen anymore. I even added OTA (which was causing other memory issues before) without any problem.

For anyone arriving here with the same issue, the way to replace the original library with the new one proposed by @touchgadget in Visual Studio Code is by adding these lines to the platformio.ini file:

```
lib_deps = 
	https://github.com/wakwak-koba/ESP32-NimBLE-Mouse.git  #master
	h2zero/NimBLE-Arduino@^1.4.0
```

Note: The repo wakwak-koba/ESP32-NimBLE-Mouse doesn´t have any releases so the only way I found to add it to VS Code is by adding the repo url (if anyone knows a better way please advice).

memetea, 2023-04-15:
Thanks for sharing. This works for me!