T-vK/ESP32-BLE-Mouse #31: Does the setBatteryLevel function actually work

lemmingDev, 2021-11-25:
Hi

Can anyone confirm that the setBatteryLevel function actually works?

On Windows, it doesn't seem to work

In the code below for example, I set the battery level to 90% when the device is started, and it shows up in Windows fine, but then I try to reduce the battery level by 1% every 30 seconds, but the information is not updated in Windows.

Any idea on how to get it working?

Thanks

```
#include <BleMouse.h>

int batteryLevel = 90;

BleMouse bleMouse("BLE Mouse", "lemmingDev", batteryLevel);

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleMouse.begin();
}

void loop() {
  if(bleMouse.isConnected()) {
    Serial.print("Battery Level: ");
    Serial.println(batteryLevel);
    bleMouse.setBatteryLevel(batteryLevel);
    delay(30000);
    batteryLevel--;
  }
}
```

lemmingDev, 2021-11-25:
[image]

T-vK, 2021-11-26:
I can confirm that it doesn't work properly. But I could never figure out why. I'm literally just calling [BLEHIDDevice::setBatteryLevel](https://github.com/espressif/arduino-esp32/blob/399f4ecbb3a4cef21e2bffa37adb6190356dfb76/libraries/BLE/src/BLEHIDDevice.cpp#L195).

https://github.com/T-vK/ESP32-BLE-Mouse/blob/8f1a15b92ff690b03a9ca45e685adc4f3a5529ad/BleMouse.cpp#L138

Maybe that's not enough or it's buggy.