T-vK/ESP32-BLE-Mouse #32: A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?

leCandas, 2021-12-27:
does not work for c3

T-vK, 2021-12-28:
Does the keyboard library work for you? https://github.com/T-vK/ESP32-BLE-Keyboard/issues/133

leCandas, 2021-12-28:
I can try it will let you know
I am using M5STAMP C3 btw

tumuyan, 2021-12-30:
esp32-c3 could pair success to my phone, but it could not link in my phone.

I just use the sample

```

#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    Serial.println("wait 10s");
    delay(10000);

    Serial.println("Sending 'Hello world'...");
    bleKeyboard.print("Hello world");

    delay(1000);

    Serial.println("Sending Enter key...");
    bleKeyboard.write(KEY_RETURN);

    delay(1000);
    
    delay(100);
    bleKeyboard.releaseAll();

  }
  Serial.println("Waiting 5 seconds...");
  delay(5000);
}

```

That's its log

```

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
entry 0x403ce000
Starting BLE work!
[   400][W][BLEService.cpp:229] addCharacteristic(): << Adding a new characteristic with the same UUID as a previous one
[   401][W][BLEService.cpp:229] addCharacteristic(): << Adding a new characteristic with the same UUID as a previous one
Waiting 5 seconds...
Waiting 5 seconds...
Waiting 5 seconds...
Waiting 5 seconds...
E (18830) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (18830) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (18836) BT_BTM: btm_proc_smp_cback received for unknown device
E (19221) BT_BTM: BTM_GetSecurityFlags false

E (19236) BT_BTM: BTM_GetSecurityFlags false

E (19251) BT_BTM: BTM_GetSecurityFlags false

E (19266) BT_BTM: BTM_GetSecurityFlags false

E (19296) BT_BTM: BTM_GetSecurityFlags false

E (19341) BT_BTM: BTM_GetSecurityFlags false

E (19371) BT_BTM: BTM_GetSecurityFlags false

E (19386) BT_BTM: BTM_GetSecurityFlags false

E (19401) BT_BTM: BTM_GetSecurityFlags false

E (19431) BT_BTM: BTM_GetSecurityFlags false

E (19446) BT_BTM: BTM_GetSecurityFlags false

E (19461) BT_BTM: BTM_GetSecurityFlags false

E (19491) BT_BTM: BTM_GetSecurityFlags false

E (19506) BT_BTM: BTM_GetSecurityFlags false

E (19521) BT_BTM: BTM_GetSecurityFlags false

E (19536) BT_BTM: BTM_GetSecurityFlags false

E (19551) BT_BTM: BTM_GetSecurityFlags false

E (19626) BT_BTM: BTM_GetSecurityFlags false

E (19641) BT_BTM: BTM_GetSecurityFlags false

E (19656) BT_BTM: BTM_GetSecurityFlags false

wait 10s
Sending 'Hello world'...
Sending Enter key...
Waiting 5 seconds...
wait 10s
Sending 'Hello world'...
Sending Enter key...
Waiting 5 seconds...
wait 10s
Sending 'Hello world'...

```

LIN1JIARONG, 2022-03-28:
I can not understand.

LIN1JIARONG, 2022-03-29:
I solve this problem just now.I find I did not change the chip type to ESP32C3.Because I new a project,the default chip type is ESP32
[image]
.