T-vK/ESP32-BLE-Mouse #37: how to reconnect the ble-mouse when device restarted？

jellybins, 2022-04-07:
when  one device connected the ble-mouse successfully，if i restart the device，it can not reconnect  again and can not find the ble-mouse by using any other device at the same time， except restart the ble-mouse，3q！

jellybins, 2022-04-07:
i see that only one device can be connected to the mouse at a time，but when i restart the one device（has  connected the ble-mouse） it cannot reconnect  again。 and i tried to find the ble-mouse failed use any other devices at that time .

jellybins, 2022-04-15:
i resolve  the problem by another way(when disconnect  more than 5 times soft restart the dev board)；and i saw the bleMouse.end() is an empty func do nothing.

ywz978020607, 2022-06-29:
> 

But how to disconnect all devices in esp32? Is there any api i am missing?

ywz978020607, 2022-06-29:
> i resolve the problem by another way(when disconnect more than 5 times soft restart the dev board)；and i saw the bleMouse.end() is an empty func do nothing.

Wow, but how to soft restart? Could you show the code? soft restart will clear the device's info? When I restart the board, but the board will connect to PC automatically, but i want to disconnect by esp32 through pressing a button..

jellybins, 2022-06-30:
> Wow, but how to soft restart? Could you show the code? soft restart will clear the device's info? When I restart the board, but the board will connect to PC automatically, but i want to disconnect [...]

you should create a new issue，We're not talking about the same problem

jellybins, 2025-04-17:
@T-vK just connect to the bluetooth of a telephone，then close the phone‘s bluetooth ，then start， blemouse can not  connect  to the phone again except restart esp32 dev board。