T-vK/ESP32-BLE-Mouse #43: My screen is 2K, and I set mobile blemouse move(200, 0);The actual position is incorrect.

bbhxwl, 2022-09-08:
I set 100, and he moves to the right. I set 200, and he moves to the left. Why?

T-vK, 2022-09-08:
Because a mouse is neither aware of your screen resolution, nor aware of your current mouse position. It doesn't even know the amount of pixels it will move for each unit because that is configured on the OS level. 

[x and y are signed chars which means you can't go beyond 127.](https://www.geeksforgeeks.org/maximum-value-of-signed-char-in-c/) If you do it anyway you'll get unexpected behavior as explained in the link. If you want to move the cursor further, you need to call the move method multiple times.

bbhxwl, 2022-09-09:
> 

I see. You can only enter 127 at most