T-vK/ESP32-BLE-Mouse #47: Good suggestion for developers use wifi

okbach, 2023-02-13:
Wi-Fi is faster than Bluetooth, the response time in Bluetooth is 60ms, but Wi-Fi is 20ms, 3 times faster
If wifi is supported then the library can be used more widely with other chips, like esp8266, which is cheaper than esp32, but it doesn't have bluetooth.

fish1968, 2023-03-19:
Is it possible to use Wi-Fi to connect HID devices to a computer / phone? That seems quite rare. Do you have any example demo code you've seen before? I am quite curious.

okbach, 2023-03-29:
It is true that this is a rare topic, and this page presents the same problem
One of the problems I think is that the company's system has become very restrictive, everything related to HID Human interface devices
under the pretext of security reasons
I think that the beginning will be by making a virtual driver program based on a Wi-Fi network, where the formatting of the chart receives the movement of the mouse or simulates a pressure
At the moment I don't have the full code but I do have links that might help get you started .
https://learn.microsoft.com/en-us/windows-hardware/drivers/hid/
https://github.com/BestBurning/kmclassdll

okbach, 2023-05-11:
Unfortunately, I got my idea wrong, and I discovered that the wireless mouse in the market depends on a load with it, I cut USB WiFi and an additional circuit to simulate the wireless mouse, and not just a regular WiFi
Even wireless headphones are the same technology, because systems and computers do not allow human input technology except through Bluetooth.

okbach, 2023-05-11:
You simply need a circuit
  built-in USB communication
  It communicates with the computer to simulate clicks