T-vK/ESP32-BLE-Mouse #49: ESP32-S3 Support

dejavu1987, 2023-04-04:
I haven't been able to make my ESP32-s3 work with BLE, In verbose mode, it logged following
`E (15363) BT_GATT: GATT_INSUF_AUTHENTICATION`
The thing I am not sure about is if ESP32-S3 is even supported yet.

danielturus, 2023-07-04:
Hello!

Any updated on this?

Thanks!

pavloshargan, 2023-10-18:
@danielturus @dejavu1987 

check out my fork. Now this repo works with esp32 s3
I've created a PR as well, so maybe it will be available within this repo soon

the update:
https://github.com/pavloshargan/ESP32-BLE-Mouse/tree/master

SmuleFlash, 2024-01-13:
I got E (31852) BT_GATT: GATT_INSUF_ENCRYPTION
What the changes in your code? Only see you delete empty line and put it back.

Ollie-Boyd, 2024-01-29:
@pavloshargan is it possible your changes didn't commit properly?