T-vK/ESP32-BLE-Mouse #57: Stop advertising after the disconnect

windmaomao, 2023-09-26:
It occurred to me, sometimes the mouse gets disconnected. And afterwards we can't initialize the connection anymore. So I dug into the code by writing another bluetooth server. So can I suggest to add the following lines to the disconnect code.

```
  void onDisconnect(BLEServer *pServer)
  {
    deviceConnected = false;

    BLEAdvertising *pAdvertising;
    pAdvertising = pServer->getAdvertising();
    pAdvertising->start();
  }
```

In our repo, it'll be inside `BleConnectionStatus`. I can raise a PR if you think this is not a bad idea, I can even add a flag to enable/disable it.

alextrical, 2023-10-30:
Hi, I'm having the same issue as you have mentioned here. However I'm struggling to see how to apply the fixed code that you have quoted.

the contents of `BleConnectionStatus.cpp` are the following
```
#include "BleConnectionStatus.h"

BleConnectionStatus::BleConnectionStatus(void) {
}

void BleConnectionStatus::onConnect(BLEServer* pServer)
{
  this->connected = true;
  BLE2902* desc = (BLE2902*)this->inputMouse->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(true);
}

void BleConnectionStatus::onDisconnect(BLEServer* pServer)
{
  this->connected = false;
  BLE2902* desc = (BLE2902*)this->inputMouse->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(false);
}
```

windmaomao, 2023-10-30:
i created this PR to address this, and I have used this for a while, it's quite stable after this change.

https://github.com/T-vK/ESP32-BLE-Mouse/pull/62/files

alextrical, 2023-10-31:
Awesome, thank you for clarifying. I think i was close but was missing the final piece to get it working reliably.

Its also helped me solve connecting one "Mouse" to multiple clients, by appending the same snippet into the `onConnect` block

coofercat, 2024-04-29:
I just tried your patch, and also found things work much better this way - random and intentional disconnects all seem to be handled gracefully with it in place.