T-vK/ESP32-BLE-Mouse #58: It works on Android, but the mouse does not work in Windows 10 and 11 environments

EliseEgkart, 2023-10-10:
Numerous tests have shown that it works on Android, but the mouse does not work in Windows 10 and 11 environments. Both Android and Windows work well with the BLE keyboard. I haven't found anyone who has a similar problem with me. Can you help me?

schnitzelguy, 2023-11-11:
Hey, I've got the same problem. I am able to pair the ESP to Windows and it knows that it's supposed to be a mouse, but somehow doesn't put it into the peripherial devices and doesn't register movements. There are no problems in Android.

[image]

schnitzelguy, 2023-11-11:
It seems the problem only occures on some dev boards. My TinyPico doesn't work, but another AZ-Delivery board does

technohelp, 2023-12-01:
I'm having the exact same problem. @schnitzelguy were you able to get it to work?

EliseEgkart, 2023-12-02:
@technohelp  https://github.com/alexz006/ESP32-BLE-Combo
Try this link's library.
When I used this library, It was the only library in GitHub that could work on the mouse
If you don't like the recognition rate when you use the combo library
I recommend dividing the keyboard and mouse functions between cores into RTOS.

I didn't like the recognition rate. Finally, using the mouse keyboard control source from Microsoft's DotNet framework 
Serial communication data was translated and used in the computer with Bluetooth Classic communication using SPP virtual port

Below is my YouTube.
https://www.youtube.com/watch?v=O_tnOcsPOvQ