T-vK/ESP32-BLE-Mouse #65: W10 not working

1PIayer, 2024-01-15:
Same example code tried with android(it works) and windows10(not works). Have i need to install windows app or driver?

OtarieSupreme, 2024-02-05:
Same here, work on Android but not on Windows 10

santhu8055, 2024-02-07:
Its working properly on windows 11 but not on 10 and 7

santhu8055, 2024-02-08:
> Same example code tried with android(it works) and windows10(not works). Have i need to install windows app or driver?

Update windows ,it will work 100%

santhu8055, 2024-02-08:
> Same here, work on Android but not on Windows 10

Update windows ,it will work 100%

OtarieSupreme, 2024-02-10:
Unfortunately, I'm stuck with Windows 10
Do you know more about the issue ? I'd like to find a workaround

OtarieSupreme, 2024-02-10:
I switched to https://github.com/wakwak-koba/ESP32-NimBLE-Mouse/tree/master and it works on Win10

santhu8055, 2024-02-13:
Dose it work only on Windows 10 or is it compatible with all versions of Windows