T-vK/ESP32-BLE-Mouse #69: ESP32 C3 Super Mini Issue

Roger2603, 2024-05-08:
I'm encountering a peculiar issue with my project involving BLEMouse integration. Interestingly, the code works perfectly fine on an ESP32 board, but when I switch to an ESP32 C3 board, I encounter a problem where the code stops executing after calling `bleMouse.begin();` in the `void setup()` function.

I've thoroughly checked the hardware connections and confirmed that they are correct. Additionally, the BLEMouse library is compatible with both ESP32 and ESP32 C3 boards. Despite these efforts, I'm unable to pinpoint the exact cause of the problem on the ESP32 C3 board.

Has anyone else experienced similar issues when transitioning between different ESP32 boards? Any suggestions or insights into potential compatibility issues between the BLEMouse library and the ESP32 C3 board would be greatly appreciated.

Thank you for your help!

alexkoppelman, 2025-02-10:
Same problem here. Any ideas anyone?

alexkoppelman, 2025-02-10:
> Same problem here. Any ideas anyone?

ok, I found you can actually make this work if you downgrade the ESP32 by Espressif to 2.0.17 in the Board manager. The 3.X's seem to have issues.

mamawhes, 2025-04-11:
@alexkoppelman api problem