T-vK/ESP32-BLE-Mouse #70: Exit status 1, shows a bunch of errors in BLE library

CDLXXVII, 2024-06-06:
Tried to compile an example, it took way to long and in the end exited with the following:
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Mouse\BleMouse.cpp: In static member function 'static void BleMouse::taskServer(void*)':
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Mouse\BleMouse.cpp:143:37: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  143 |   BLEDevice::init(bleMouseInstance->deviceName);
      |                   ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
      |                                     |
      |                                     std::string {aka std::__cxx11::basic_string<char>}
In file included from c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Mouse\BleMouse.cpp:1:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLEDevice.h:41:27: note:   initializing argument 1 of 'static void BLEDevice::init(String)'
   41 |   static void init(String deviceName);                                                            // Initialize the local BLE environment.
      |                    ~~~~~~~^~~~~~~~~~
c:\Users\Admin\Documents\Arduino\libraries\ESP32_BLE_Mouse\BleMouse.cpp:151:50: error: no matching function for call to 'BLECharacteristic::setValue(std::string&)'
  151 |   bleMouseInstance->hid->manufacturer()->setValue(bleMouseInstance->deviceManufacturer);
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLEServer.h:23,
                 from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLEDevice.h:21:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:77:8: note: candidate: 'void BLECharacteristic::setValue(uint8_t*, size_t)'
   77 |   void setValue(uint8_t *data, size_t size);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:77:8: note:   candidate expects 2 arguments, 1 provided
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:78:8: note: candidate: 'void BLECharacteristic::setValue(String)'
   78 |   void setValue(String value);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:78:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
   78 |   void setValue(String value);
      |                 ~~~~~~~^~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:79:8: note: candidate: 'void BLECharacteristic::setValue(uint16_t&)'
   79 |   void setValue(uint16_t &data16);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:79:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint16_t&' {aka 'short unsigned int&'}
   79 |   void setValue(uint16_t &data16);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:80:8: note: candidate: 'void BLECharacteristic::setValue(uint32_t&)'
   80 |   void setValue(uint32_t &data32);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:80:27: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'uint32_t&' {aka 'long unsigned int&'}
   80 |   void setValue(uint32_t &data32);
      |                 ~~~~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:81:8: note: candidate: 'void BLECharacteristic::setValue(int&)'
   81 |   void setValue(int &data32);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:81:22: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'int&'
   81 |   void setValue(int &data32);
      |                 ~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:82:8: note: candidate: 'void BLECharacteristic::setValue(float&)'
   82 |   void setValue(float &data32);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:82:24: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'float&'
   82 |   void setValue(float &data32);
      |                 ~~~~~~~^~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:83:8: note: candidate: 'void BLECharacteristic::setValue(double&)'
   83 |   void setValue(double &data64);
      |        ^~~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE\src/BLECharacteristic.h:83:25: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'double&'
   83 |   void setValue(double &data64);
      |                 ~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

CuldesacDude, 2024-06-18:
Had a similar problem after the new arduino update here is what fixed my problem in the file BleMouse.cpp, esseantiall just need to change 'string' into String. Just copy this and replace the function.

`void BleMouse::taskServer(void* pvParameter) {
  BleMouse* bleMouseInstance = (BleMouse *) pvParameter; //static_cast<BleMouse *>(pvParameter);
  BLEDevice::init(String(bleMouseInstance->deviceName.c_str())); // Line 143 modified   <<<<<<
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(bleMouseInstance->connectionStatus);

  bleMouseInstance->hid = new BLEHIDDevice(pServer);
  bleMouseInstance->inputMouse = bleMouseInstance->hid->inputReport(0); // <-- input REPORTID from report map
  bleMouseInstance->connectionStatus->inputMouse = bleMouseInstance->inputMouse;

  bleMouseInstance->hid->manufacturer()->setValue(String(bleMouseInstance->deviceManufacturer.c_str())); // Line 151 modified <<<<<<

  bleMouseInstance->hid->pnp(0x02, 0xe502, 0xa111, 0x0210);
  bleMouseInstance->hid->hidInfo(0x00,0x02);

  BLESecurity *pSecurity = new BLESecurity();

  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

  bleMouseInstance->hid->reportMap((uint8_t*)_hidReportDescriptor, sizeof(_hidReportDescriptor));
  bleMouseInstance->hid->startServices();

  bleMouseInstance->onStarted(pServer);

  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->setAppearance(HID_MOUSE);
  pAdvertising->addServiceUUID(bleMouseInstance->hid->hidService()->getUUID());
  pAdvertising->start();
  bleMouseInstance->hid->setBatteryLevel(bleMouseInstance->batteryLevel);

  ESP_LOGD(LOG_TAG, "Advertising started!");
  vTaskDelay(portMAX_DELAY); //delay(portMAX_DELAY);
}
`

WilliamHVollita, 2024-12-17:
> Had a similar problem after the new arduino update here is what fixed my problem in the file BleMouse.cpp, esseantiall just need to change 'string' into String. Just copy this and replace the [...]

cooooool bro! just fixed my problem. thx so much!

robeeeert, 2025-01-01:
Thank you @CuldesacDude ! I needed a second similar adjustment in line 143:

BLEDevice::init(String(bleMouseInstance->deviceName.c_str()));