T-vK/ESP32-BLE-Mouse #78: Cant Get High res scrolling working

Mallo321123, 2025-05-01:
Hey, there. I am trying to implement High-Res scrolling into the library, but I, failing to do it, could someone help me please?
Here is my custom HID table:

```
static const uint8_t _hidReportDescriptor[] = {
    USAGE_PAGE(1), 0x01, // USAGE_PAGE (Generic Desktop)
    USAGE(1), 0x02,      // USAGE (Mouse)
    COLLECTION(1), 0x01, // COLLECTION (Application)
      USAGE(1), 0x01,      //   USAGE (Pointer)
      COLLECTION(1), 0x00, //   COLLECTION (Physical)
        // ------------------------------------------------- Buttons (Left, Right, Middle, Back, Forward)
        USAGE_PAGE(1), 0x09,      //     USAGE_PAGE (Button)
        USAGE_MINIMUM(1), 0x01,   //     USAGE_MINIMUM (Button 1)
        USAGE_MAXIMUM(1), 0x05,   //     USAGE_MAXIMUM (Button 5)
        LOGICAL_MINIMUM(1), 0x00, //     LOGICAL_MINIMUM (0)
        LOGICAL_MAXIMUM(1), 0x01, //     LOGICAL_MAXIMUM (1)
        REPORT_SIZE(1), 0x01,     //     REPORT_SIZE (1)
        REPORT_COUNT(1), 0x05,    //     REPORT_COUNT (5)
        HIDINPUT(1), 0x02,        //     INPUT (Data, Variable, Absolute) ;5 button bits
        // ------------------------------------------------- Padding
        REPORT_SIZE(1), 0x03,  //     REPORT_SIZE (3)
        REPORT_COUNT(1), 0x01, //     REPORT_COUNT (1)
        HIDINPUT(1), 0x03,     //     INPUT (Constant, Variable, Absolute) ;3 bit padding
        // ------------------------------------------------- X/Y position, Wheel
        USAGE_PAGE(1), 0x01,      //     USAGE_PAGE (Generic Desktop)
        USAGE(1), 0x30,           //     USAGE (X)
        USAGE(1), 0x31,           //     USAGE (Y)
        USAGE(1), 0x38,           //     USAGE (Wheel)
        LOGICAL_MINIMUM(1), 0x81, //     LOGICAL_MINIMUM (-127)
        LOGICAL_MAXIMUM(1), 0x7f, //     LOGICAL_MAXIMUM (127)
        REPORT_SIZE(1), 0x08,     //     REPORT_SIZE (8)
        REPORT_COUNT(1), 0x03,    //     REPORT_COUNT (3)
        HIDINPUT(1), 0x0c,        //     INPUT (Data, Variable, Relative) ;3 bytes (X,Y,Wheel)
        // ------------------------------------------------- Horizontal wheel
        USAGE_PAGE(1), 0x0c,      //     USAGE PAGE (Consumer Devices)
        USAGE(2), 0x38, 0x02,     //     USAGE (AC Pan)
        LOGICAL_MINIMUM(1), 0x81, //     LOGICAL_MINIMUM (-127)
        LOGICAL_MAXIMUM(1), 0x7f, //     LOGICAL_MAXIMUM (127)
        REPORT_SIZE(1), 0x08,     //     REPORT_SIZE (8)
        REPORT_COUNT(1), 0x01,    //     REPORT_COUNT (1)
        HIDINPUT(1), 0x06,        //     INPUT (Data, Var, Rel)
        // ------------------------------------------------- High-Resolution Wheel support
        USAGE_PAGE(1), 0x0C,           //     USAGE PAGE (Consumer Devices)
        USAGE(2), 0xE8, 0x03,          //     USAGE (Resolution Multiplier)
        LOGICAL_MINIMUM(1), 0x81,
        LOGICAL_MAXIMUM(1), 0x7f,
        REPORT_SIZE(1), 0x10,          //      Bits
        REPORT_COUNT(1), 0x02,
        FEATURE(1), 0x02, // FEATURE (Data, Var, Abs)
      END_COLLECTION(0),        //   END_COLLECTION
    END_COLLECTION(0)         // END_COLLECTION
};
```

Mallo321123, 2025-05-09:
Quick Update:
I found out, that the hid descriptor table is fully being ignored. I can even leave it completly empty and inputs are still working. I have given up at this point already.

EbrithilNogare, 2025-06-22:
Hi, I'm working on this as well, and I encountered the same issue with the descriptor table being ignored, but it is only because **Windows is caching it** somehow. I can't tell you more details, but if you **remove the device and add it again**, the descriptor is fetched again.

EbrithilNogare, 2025-06-22:
I got it working with a descriptor like this (I need only scrolling, and X, Y are mandator,y I think)

```c
static const uint8_t _hidReportDescriptor[] = {
  USAGE_PAGE(1),       0x01, // USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x02, // USAGE (Mouse)
  COLLECTION(1),       0x01, // COLLECTION (Application)
  USAGE_PAGE(1),       0x01, //   USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x02, //   USAGE (Mouse)
  COLLECTION(1),       0x02, //   COLLECTION (Logical)
  REPORT_ID(1),        0x01, //     REPORT_ID (1)
  USAGE(1),            0x01, //     USAGE (Pointer)
  COLLECTION(1),       0x00, //     COLLECTION (Physical)
  // ------------------------------------------------- X/Y position
  USAGE_PAGE(1),       0x01, //       USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x30, //       USAGE (X)
  USAGE(1),            0x31, //       USAGE (Y)
  LOGICAL_MINIMUM(1),  0x81, //       LOGICAL_MINIMUM (-127)
  LOGICAL_MAXIMUM(1),  0x7f, //       LOGICAL_MAXIMUM (127)
  REPORT_SIZE(1),      0x08, //       REPORT_SIZE (8)
  REPORT_COUNT(1),     0x02, //       REPORT_COUNT (2)
  HIDINPUT(1),         0x06, //       INPUT (Data, Variable, Relative) ;3 bytes (X,Y,Wheel)
  // --------------------------------------------------- Wheel
  USAGE(1),            0x38, //       USAGE (Wheel)
  PHYSICAL_MINIMUM(1), 0x00, //       PHYSICAL_MINIMUM (0)
  PHYSICAL_MAXIMUM(1), 0x00, //       PHYSICAL_MAXIMUM (0)
  LOGICAL_MINIMUM(1),  0x81, //       LOGICAL_MINIMUM (-127)
  LOGICAL_MAXIMUM(1),  0x7f, //       LOGICAL_MAXIMUM (127)
  REPORT_SIZE(1),      0x08, //       REPORT_SIZE (8)
  REPORT_COUNT(1),     0x01, //       REPORT_COUNT (1)
  HIDINPUT(1),         0x06, //       INPUT (Data, Variable, Relative)
  // ------------------------------------------------- Resolution Multiplier
  USAGE_PAGE(1),       0x01, //       Generic Desktop
  USAGE(1),            0x48, //       Usage: Resolution Multiplier
  LOGICAL_MINIMUM(1),  0x00, //       Logical Min = 0
  LOGICAL_MAXIMUM(1),  0x01, //       Logical Max = 1
  PHYSICAL_MINIMUM(1), 0x01, //       Physical Min = 1
  PHYSICAL_MAXIMUM(1), 0x80, //       Physical Max = 128
  REPORT_SIZE(1),      0x08, //       REPORT_SIZE (8)
  REPORT_COUNT(1),     0x01, //       REPORT_COUNT (1)
  FEATURE(1),          0x02, //       Feature (Data,Var,Abs)
  END_COLLECTION(0),         //     END_COLLECTION (Physical)
  END_COLLECTION(0),         //   END_COLLECTION (Logical)
  END_COLLECTION(0),         // END_COLLECTION (Application)
};
```

And setting the resolution like this

```c
bleMouseInstance->inputMouse = bleMouseInstance->hid->inputReport(0x01); // <-- input REPORTID from report map
bleMouseInstance->featureResolution = bleMouseInstance->hid->featureReport(0x01); // <-- feature REPORTID for resolution multiplier  
bleMouseInstance->featureResolution->setValue(new uint8_t{0x80}, 1);
```

Mallo321123, 2025-06-25:
Hey, Thanks for your reply.
The Caching tip was very helpfull. (I am using Linux btw, but its the same)

I have a "Problem" with these two Lines, there is no `featureResolution`, for now I just ignored these
```
bleMouseInstance->featureResolution = bleMouseInstance->hid->featureReport(0x01); // <-- feature REPORTID for resolution multiplier  
bleMouseInstance->featureResolution->setValue(new uint8_t{0x80}, 1);
```

The report map is getting recognized correctly I think. Here is what the evtest result looks like:
```
Supported events:
  Event type 0 (EV_SYN)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
```

What does your move function look like?
Right now I am not recieving move commands in evtest

leollo98, 2025-06-26:
try verifying by libinput with
`sudo libinput debug-events --verbose`

once you send the command move only the scroll whell and see what is shown, my for example appears 
```
 event24  POINTER_SCROLL_WHEEL    +1.170s       vert 0.38/3.0* horiz 0.00/0.0 (wheel)
 event24  POINTER_SCROLL_WHEEL    +1.170s       vert 0.38/3.0* horiz 0.00/0.0 (wheel)
 event24  POINTER_SCROLL_WHEEL    +1.170s       vert 0.25/2.0* horiz 0.00/0.0 (wheel)
 event24  POINTER_SCROLL_WHEEL    +1.215s       vert 0.12/1.0* horiz 0.00/0.0 (wheel)
```

Mallo321123, 2025-06-26:
There are no scroll move commands being recognized.
Normal X and Y movements are displayed.

The device is getting recognized correctly:
```-event258  DEVICE_ADDED            ESP32 Bluetooth Mouse             seat0 default group9  cap:kp scroll-nat```

leollo98, 2025-06-27:
I'm going to fork the repo and create a new library based on this one on the next few days with buttons and the resolution multiplayer, so you can use it

leollo98, 2025-06-28:
I forked with the changes at https://github.com/leollo98/ESP32-BLE-Mouse or at platformIO with the name leollo98/ESP32 BLE Mouse With Precision Scroll

leollo98, 2025-06-29:
I'm using Kubuntu 24.04, with libinput 1.25.0

check if you need to add a input at `/etc/udev/hwdb.d/`, I added a file with name `11-esp32-mouse.hwdb` and content:

```
mouse:*:name:ESP*:
    MOUSE_WHEEL_CLICK_ANGLE=7.5
    MOUSE_WHEEL_CLICK_COUNT=120
```

reboot after.

For some reason Firefox and it's forks, don't support precision scrolling, but you can check it on chrome and it's forks

closed 2025-10-22 by Mallo321123